NorthernWidget-Skunkworks/Project-Walrus #3: Switch I2C Output Connections

bschulz1701, 2019-08-07:
### Problem:
External I2C connections (into cable) are not consistent with others using the same cable ([Project Haar](https://github.com/NorthernWidget-Skunkworks/Project-Haar))

### Potential Solution:
Change the SDA/SCL configuration to match Haar

awickert, 2021-02-13:
@bschulz1701: I believe that this cable design is now consistent with our other designs.

However, if we use the AlphaWire cable, the pinout wire colors are unintuitive. I believe that we're approaching the go-or-no-go moment for deciding whether to change all HardMount pinouts (for now: this and https://github.com/NorthernWidget-Skunkworks/Project-Haar). My vote would be:
* YES to change
* to increment the version numbers to reflect this change
* To use the Brother labelmaker to include wraparound able labels with the pinout noted; these should be placed at a far enough distance from the wires to the loggers that they do not interfere with the cable gland.

What is your opinion here?

awickert, 2021-02-13:
A bit more information:

**Current pinout:**
| **Color** | **Connection** |
|-----------|----------------|
| Brown     | V+ (3.3-5.5 V) |
| White     | GND            |
| Black     | SDA            |
| Blue      | SCL            |

**Proposed pinout:**
| **Color** | **Connection** |
|-----------|----------------|
| White     | V+ (3.3-5.5 V) |
| Black     | GND            |
| Brown     | SDA            |
| Blue      | SCL            |

Linked issue in Haar repository: https://github.com/NorthernWidget-Skunkworks/Project-Haar/issues/6