NorthernWidget-Skunkworks/Project-Walrus #4: 3D print enclosure for the HardMount design

awickert, 2020-04-25:
I know that @bschulz1701 had been working on a HardMount enclosure but was having difficulties with the epoxy pour-over because of the height of the plug vs. the height of the pressure transducer. We then discussed a screw-on enclosure with a conformal coating and/or just using wires soldered to the SMD pads. What is our final plan? And if there is any CAD work to be done, I sadly don't know SolidWorks, so would have to ask @bschulz1701 or @superchap123 to do it.

superchap123, 2020-06-19:
Rev A hard mount assembly put into documentation (maybe not the best place for this, also used GitHub desktop for this and brought over some unintended files it seems, will fix later).  Design is for use with T4132412041-000 connector or like TE M12 panel mount connector. Could use some refinements/additions to the models.  Need to add 2-56 screw to assembly.  Add an ISP pin protector.  Not sure about the current dimensions for capturing the M16 panel nut, thinking it would be difficult to press on in the current state but might be OK.  Real-world tests would be helpful.  @bschulz1701 or @awickert evaluate if able.   
[image]

awickert, 2020-06-19:
@superchap123 I did a little happy dance around the cabin when I saw this. It looks fantastic and is now in the README.

I have a couple thoughts and will open other "Issues" to discuss.

awickert, 2020-06-19:
Reopening this with a question for @superchap123 or somebody: How do we get the epoxy to the bottom of the board in the tray?

superchap123, 2020-06-19:
I had started the design with the intension of clipping onto the board and the screw hole would have done the job of getting the potting compound to the bottom side but I started leaning in favor of the screw again.  My thought was to round off the end of the tray more or less matching the dome profile of the 3D printed housing which would leave a gap at the end but I had not made those revisions yet.  I am not sure how well the stuff flows and maybe the current gap would be enough but probably not and increasing the length a bit to make this easier seems worth it.  Since it is not going to fit through an M16x1.5 opening we have some room to grow in the width if we would rather as well.  I also should need to change the screw protrusion on the bottom side into something better suited for printing.  Currently, this will need to print standing up and not sure how well that will go.  Might need to go thicker.

awickert, 2020-06-20:
I was thinking of the screw hole too but then saw that you were asking about a 2-56 screw!

Bobby's standing-up prints on the Prusa have gone pretty well, but yeah, a little extra thickness can't hurt.

Sounds like you have the right ideas in mind. I plan to program a Walrus soon, and then will drop it off with you for testing and sizing.

superchap123, 2020-06-22:
I think I am going to remake the models in Onset which can be used for free but all models will be released under an open source licence.  It should also be very easy to collaborate and access the files as it is browser based software.

Chad Sandell

awickert, 2020-06-22:
You mean Onshape?
https://www.onshape.com/products/free
Sure; just send the link! Maybe I'll finally learn some 3D CAD.

superchap123, 2020-07-04:
Happy 4th!  Finally got the models remade and updated in Onshape.  I need to get the physical parts from you and start printing stuff.

Chad Sandell

https://www.onshape.com/products/free

awickert, 2020-07-04:
Excellent news and happy 4th! I'll get in touch about swinging by and dropping off the boards.

awickert, 2021-02-13:
@superchap123 completed his updates, and I have just updated the repo to include the 3D print and full assembly files.
15a6ea1ceea464b84c4a1b21669ceb5cb8cb5855

closed 2021-02-13 by awickert